bluerobotics/ardusub-zola #1: Enable Version Switching

ES-Alexander, 2021-10-18:
Bringing together https://github.com/bluerobotics/ardusub-gitbook/issues/227 and https://github.com/bluerobotics/ardusub-gitbook/issues/229, each software will be documented in its own set of branches (by semi-major version, e.g. `ardusub-4.0`), with tags used for minor versions (`ardusub-4.0.3`), and submodules used within the `main` branch site content to link to each version. A suitable switching mechanism will need to be made, which should be helped by Zola's ability to apply things to all pages in a given section.

ES-Alexander, 2021-10-18:
While branches for `companion`, `ardusub`, and `qgroundcontrol` are a given/requirement, it may also be worth having a branch that allows selecting Python version for the `pymavlink` examples.

ES-Alexander, 2021-11-16:
Need to add a macro to the header of versioned sections/pages that does something along the lines of
```html
<script>
  function change_version() {
    ... // modify url by replacing version number
  } 
</script>

<select name="versions" id="versions" onChange="change_version()">
  {% for version in versions %}
    <option value={{ version }}>{{ version.title }}</option>
  {% endfor %}
</select>
```

- `versions` needs to be determined in a way that allows nested pages to still change versions (e.g. `software/companion/1.0/configuration/...` should be able to go to `software/companion/1.0.1/configuration/...`) - perhaps check path parent until a non-versioned parent section of the current versioned section is found
- should have an intelligent fallback if an equivalent page doesn't exist (e.g. go to root of version)
- searching is tricky - would be ideal to have options for general searching over the full docs (preferentially indexed to the latest version of each software, somehow), and another option for searching specifically within the selected version of the current software docs (based on the current page)
- @patrickelectric has suggested it may be possible to do the `change_version` page-changing functionality using just Zola constructs/functions rather than using javascript

ES-Alexander, 2021-11-18:
May be relevant to consider Rust's version selector:

[image]

rafaellehmkuhl, 2022-02-10:
> While branches for `companion`, `ardusub`, and `qgroundcontrol` are a given/requirement, it may also be worth having a branch that allows selecting Python version for the `pymavlink` examples.

If we really need different python versions, I would use tabs on the code visualizer.

ES-Alexander, 2022-02-11:
> If we really need different python versions, I would use tabs on the code visualizer.

No idea if that's possible, but it does seem like a nice way of handling it, particularly given it's a different kind of version than the others (version of underlying technology, rather than version of the software/library itself).

closed 2024-11-15 by ES-Alexander