bluerobotics/ardusub-zola #72: we are missing "build instructions"

Williangalvani, 2023-04-08:
I did:
`sudo pacman -S zola`
`git submodule update --init --recursive`
`zola serve -i 0.0.0.0` (testing from another computer)

But I apparently got no css. am I missing something?

ES-Alexander, 2023-04-13:
I would expect such instructions to be covered by #2, so I think this is a duplicate issue?

> I did:
> ...
> But I apparently got no css. am I missing something?

I generally `zola serve` without `-i`, and I installed zola using `brew`, but otherwise the same and it works fine for me. Note that building/serving should be done from the root of the `master` branch - if you're in a different directory or branch then it likely won't work. 

Are you saying the generated site wasn't formatted, or that you specifically couldn't find any css files in the source? Adidoks (the theme we're using) uses [Sass](https://sass-lang.com/documentation/) stylesheets, which are `.scss` files, if that's relevant.

Williangalvani, 2023-04-13:
> I generally `zola serve` without `-i`, and I installed zola using `brew`, but otherwise the same and it works fine for me. Note that building/serving should be done from the root of the `master` branch - if you're in a different directory or branch then it likely won't work.

This is the issue, I was on a branch based on your PR

ES-Alexander, 2024-02-22:
More detailed process [here](https://github.com/bluerobotics/cockpit/pull/747#issuecomment-1958741434)

ES-Alexander, 2024-10-16:
[Provided in new contributing docs](https://github.com/bluerobotics/BlueOS-docs/blob/latest/CONTRIBUTING.md#local-setup)

closed 2024-10-16 by ES-Alexander